Seattle-Aquarium/CCR_development #1: Enhancements for KelpNet Based on CoralNet-Toolbox

m-h-williams, 2024-06-07:
**Description:**
CoralNet-Toolbox has opened our eyes to the possibilities of what KelpNet could be. We are very excited about the opportunity to expand upon this toolbox and build an open-access image classification and annotation platform for temperate benthic species.

**Tools and Suggested Improvements of CoralNet-Toolbox for Our Application:**

Patches Tool
- What We Like: It makes it really easy to gather a large number of image patches for training a classifier.
- Possible Improvements: Changing the color of the boxes when you change the base name (i.e., the label) would make visualization of the patches for discrete labels easier.

Documentation/Ease of Use:
- Required Software: Provide details about the required software to use the toolbox, e.g., Python IDE, CoralNet user information, etc.
- Information Tabs: Include an information tab for each tool that details what the tool is and the required files.
- User Accessibility: Using the Toolbox requires at least some Python knowledge, which can limit the number of possible users. Is there a workaround to make it more accessible?

General Issues:
- Tool Loading Issue: After a period of time, a tool page that was opening when the Toolbox was first opened may not open again. The info pop-up will say the tool is opening, then the pop-up will close, and nothing will have opened. This can occasionally be fixed by restarting the Python kernel. However, some pages/tools, like the demos for image classification and SAM, never seem to load.

clydemcqueen, 2024-06-26:
For reference, here is the CoralNet-Toolbox source code: https://github.com/Jordan-Pierce/CoralNet-Toolbox

zhrandell, 2025-04-12:
Sabrina has been leading the charge on a fork of CoralNet-Toolbox; she has developed the ability to review and select image patches to delete. Some of these details can be tracked at the following CoralNet-Toolbox issue: 

https://github.com/Jordan-Pierce/CoralNet-Toolbox/issues/25

m-h-williams, 2025-07-17:
Jordan has added a lot of functionality to CoralNet-Toolbox that meets and surpasses our asks.. Allowing us create a custom classifier, patch editing, and object tracking.