Seattle-Aquarium/CCR_development #8: Sensor suite diagrams and reference frame descriptions

tonioteran, 2024-09-05:
It would be nice to have a figure that pictorially describes both our sensor suites (the vessel side, and the ROV side). Something like "Figure 2" on [this paper](https://furgalep.github.io/sbib/furgale_ijrr12.pdf) would be nice (adding a screenshot just for easy referencing).

[image]

This will also help us in making sure that we have every reference frame calibrated and accounted for during our data processing steps.

Happy to provide an initial attempt at this if you point me to the right place haha, otherwise I can provide assistance in crafting it.

zhrandell, 2024-09-05:
Great suggestion, thanks @tonioteran! One option that I'd be more than happy to pursue would be looping in the Seattle Aquarium's Marketing folks . . . we have people with some pretty slick graphic design skills, and I think with the appropriate imagery, they could whip up a figure for us relatively easily. Such a fig with the vessel (including GNSS compass, command console, G2 box, USBL antenna), and ROV (and all associated sensors).

I'll ping our Marketing folks to see if they have bandwidth to do something like this (I'm hoping so, now that the new building is open). If they're game, I'll generate a draft sketch, bounce that off you to see if it conceptually captures what's required, then I can pass that off to them (with photos of the vessel, sensors, etc.). I'll also make sure we're able to easily add text to such a figure. 

What do you think?

zhrandell, 2024-09-05:
Green light from our Marketing graphics design folks to work with us on this visual! The main individual, Mike, is out on vacation right now, but I'll link up with him once he's back to get the ball rolling. It may take a couple weeks as there's still a substantial amount of Ocean Pavilion related signage tasks that need to be completed. But, it sounds like he'll likely be excited to work on something that isn't a sign. 

I'll work to obtain photos of all the relevent parts so that he has what he needs. 

See the attached for an example of a graphic he quickly whipped up for us for a different project. 

Finally, it occurs to me that this visual (or a version of it) would likely be very useful to include on the first scientific publication about our ROV methods :-)

[SEAQ+ROV+illustrarion+1-v1.pdf](https://github.com/user-attachments/files/16897292/SEAQ%2BROV%2Billustrarion%2B1-v1.pdf)

tonioteran, 2024-09-05:
Dang, that sounds pretty sweet and professional hahaha. 

I think to start with, though, let's do something more primitive so that we can easily iterate upon, it could even be hand drawn and stuff. Once we've settled on a good representation, we can then pass that over to marketing and have them make it really pretty?

It might even be a good idea to try and get a CAD started to get a rough outline of the relative poses of all the sensors?

zhrandell, 2024-09-05:
That sounds good to me, thanks Tonio! I'll sketch out the relative positioning and will attach photos of all our sensors on the vessel and the ROV. I'm afraid I don't have any fluency in CAD, but if you want to take those sketches and then run with it in CAD, you'd be more than welcome! I assume you'll require measurements of distances between, e.g., satellite compass and the USBL antenna, etc.? (which we have). 

We can then have the marketing folks do something more polished for presentations / manuscripts. I'll post rough sketches and images of the vessel, sensors, & ROV here on this Issue once I have them (I'll get some of that later today). 

Oh and speaking of CAD, there are indeed 3D models available of the ROV and various sensors that could be relevant to this diagram: 

- See [here](https://bluerobotics.com/store/rov/bluerov2/) on the BR website for the base BlueROV2, BUT ignore the 3D files on this page because . . . 
- . . . we use the Heavy Configuration Upgrade Kit (see [here](https://bluerobotics.com/store/rov/bluerov2-upgrade-kits/brov2-heavy-retrofit/)), with 3D models available [here](https://cad.bluerobotics.com/BROV2-HEAVY-FILES.zip).
- We also use the Payload Skid to mount the DVL and cameras (linked [here](https://bluerobotics.com/store/rov/bluerov2-accessories/brov-payload-skid/)) and see [here](https://cad.bluerobotics.com/BROV2_PAYLOAD_SKID_BR-100233.zip) for the 3D files.
- See [this](https://waterlinked.com/dvl-a50#Downloads-%2F-Resources%E2%80%8B) WaterLinked page for 3D models of the DVL and mounting bracket. 
- A CAD file for the U1 transmitter (the transmitter on the ROV for our USBL system) can be found [here](https://waterlinked.com/web/content/7587?unique=bf69e90278cd8e6aa5a8d6171493bbd4a5396e01&download=true).
- A 3D file for the acoustic antenna on our vessel (as part of the USBL system) can be found [here](https://waterlinked.com/web/content/7666?unique=ee7b6084fd2b3e1e4532223d99b153717576efe5&download=true).
- The GNSS Compass we use on the vessel to provide GPS / compass data can be found [here](https://www.advancednavigation.com/inertial-navigation-systems/satellite-compass/gnss-compass/), and a 3D model is available [here](https://www.advancednavigation.com/wp-content/uploads/2021/08/GNSS-Compass-Ethernet-3D-Model-STEP.zip). 
- TODO . . . find a good 3D model of the GoPro underwater housing.

clydemcqueen, 2024-09-05:
We could also provide a Gazebo models for Lutris and Nereo for researchers. These could have the exact frames & measurements in code, and could be used for simulations.

Patrick Pereira (BR) started a model a long while back:
https://github.com/patrickelectric/bluerov_ros_playground

I ported this to Gazebo Garden. Then Evan Palmer (OSU) added a heavy version:
https://github.com/clydemcqueen/bluerov2_ignition/tree/main/models

Evan has some newer models, I have not looked at these but they might be helpful:
https://github.com/Robotic-Decision-Making-Lab/blue/tree/main/blue_description/description

tonioteran, 2024-09-06:
honestly, im not really good at CADing much, but i can give it a shot; might be interesting especially connected to the gazebo models that @clydemcqueen points out

zhrandell, 2024-09-06:
@tonioteran no pressure to CAD if that isn't part of the workflow . . . I'm not fully up to speed on what will be required here, so if CAD'ing is useful, great, but if not, no pressure.  

I've started documenting some of the sensors, particularly those on the vessel. I'm going to gather photos here on [this DropBox folder](https://www.dropbox.com/scl/fo/fcmttqmj4jxumhqbc79gq/AJAZYxFrW9COdcB4fGetxRU?rlkey=5gyqqc0j3tzurv6y9rlo8lq72&dl=0) . . . you should be able to view / download them. I took measurements as well, and over the next little while I'll work to get those translated to a sketch.

zhrandell, 2024-09-07:
@tonioteran, good call on not trying to start a fancy image right away. It was actually pretty relaxing to sketch something by hand. I won't claim to be an artist, but we at least now have a starting point for a sketch of the vessel's sensors. 

Check out the folder [linked here](https://drive.google.com/drive/folders/1ayAP2tN2MUOREs8Gpf2d01ko96rzM4Xq?usp=drive_link) . . . there's a Google Slides presentation titled `sensor_and_vessel_sketch`. In it a have two sketches of the vessel. I've removed as much of the image background as possible so that we can add text, measurements, etc. 

We have a folder with the raw sketches, as well as a folder with photos of the vessel and vessel sensors. 

I did take measurements by myself at the vessel yesterday, but to make this as accurate as possible I think I should re-do the measurements on Tuesday or Wednesday this week when there are 2+ people present on the boat . . . that'll give us a pinch more precision. 

Are these sketches + the Google Slides a useful format? Is there a better way to compile this information? I'm open to ideas!

tonioteran, 2024-09-10:
nice nice nice, the sketches are super good

do you have the measurements' numbers somewhere? we can start encoding them into relative transforms between all sensors and plotting them in 3D to sanity check them and make sure we're happy with them

note that we need to also figure out relative orientation, not just the position, not sure how to best go about that haha? we might need to collect a bunch of sensor data and run some solve geometric optimization?  how have you guys been doing it?

zhrandell, 2024-09-12:
@tonioteran, I took measurements on the vessel myself last week, but doing so without a second pair of hands was challenging, and I'm not super confident in the data. So, this upcoming Tuesday, Megan and I will meet at the vessel at Bell Harbor Marina at 8am and we'll gather all the required measurements. 

Regarding relative orientation and geometric optimization . . . I don't think we've been doing any of that . . .

tonioteran, 2024-09-22:
super nice! thank you!
lemme take a look!

zhrandell, 2024-11-20:
I believe this is complete . . . @tonioteran, please hesitate to re-open this issue if anything additional is required!

closed 2024-11-20 by zhrandell