Seattle-Aquarium/CCR_development #13: Log WL G2 orientation

clydemcqueen, 2024-09-10:
Log the orientation of the WL G2 box, and / or raw IMU data.

Step 1: contact WL to see if this is possible.

clydemcqueen, 2024-09-11:
Ticket submitted, waiting for a reply.

clydemcqueen, 2024-09-12:
Scrott from WL reports that this API should do the trick: https://demo.waterlinked.com/swagger/#/imu/imu%23Get

Next steps:
* try this on the G2 box. @zhrandell can you bring a G2 box to lab day next week?
* make a plan to log this data

zhrandell, 2024-09-12:
Great news! Yes, I can bring a G2 box to the lab day. In fact, I will bring both G2 boxes. They both are sorely due for a software update, which has historically taken several hours (??). So, once you've finished interfacing via the API, we can start the update process. 

We can also have Nereo with us on lab day to enable any relevant EM testing (and we also need to update the DVL software!).

clydemcqueen, 2024-09-26:
I tried this on the G2 box, and it works.

Next challenge is to log it.

clydemcqueen, 2024-12-11:
Done. I have data from 1 dive. I don't see big evidence that the g2 box is moving around... it seems pretty stable. That's good news!

I also compared 2 solutions using ros2 tools (rviz2):
* GPS_INPUT messages, these are the wl-ugps-extension-generated "measurements."
* The GPS info from the GGA + the heading from the HDT + the acoustic solution from the g2.

To my eye these look pretty close, which is pretty great news: it means we can trust the GPS_INPUT messages.

Closing this, as we are logging. Any further analysis will be a separate issue.

closed 2024-12-11 by clydemcqueen