Seattle-Aquarium/CCR_development #14: Log GNSS compass data

clydemcqueen, 2024-09-10:
Log GNSS compass position and orientation.

[See the 1-pager for a detailed description](https://github.com/zhrandell/Seattle_Aquarium_CCR_development/blob/main/1-pagers/GNSS_compass_logging.md)

clydemcqueen, 2024-09-29:
I don't see a standard NMEA 0183 message that contains roll and pitch.

The Advanced Navigation GNSS compass [sends roll and pitch in a few proprietary formats](https://docs.advancednavigation.com/gnss-compass/Operation/NMEA%200183.htm#NMEA):
* $PASHR, NMEA 0183 proprietary message, appears to be used by a few systems, see [Hexagon/NovAtel](https://docs.novatel.com/OEM7/Content/SPAN_Logs/PASHR.htm) and [Trimble](https://receiverhelp.trimble.com/mps566-gnss/NMEA-0183messages_PASHR.html)
* $TSS1, ASCII format that does _not_ follow the NMEA 0183 standard, see [Hexagon/NovAtel](https://docs.novatel.com/OEM7/Content/SPAN_Logs/TSS1.htm)
* $PFEC,GPatt, NMEA 0183 proprietary message defined by Furuno, see [Marport/Scala](https://marport.com/doc_web/scala/topics/r-NMEAInputs.html)
* ANPP is a binary format defined by Advanced Navigation
* PGN 127257, NMEA 2000, only sent at 1Hz, not configurable, see [ContinuousWave](https://continuouswave.com/whaler/reference/PGN.html) for a description

Of these, $PASHR looks like the easiest to add to wl_ugps_external_extension, so I'll start there.

clydemcqueen, 2024-09-30:
Ready for test!

To test this, upgrade the wl_ugps_external_extension to v1.1.0-beta1 and set some custom parameters:

~~~
{
  "Env": [
    "UGPS_HOST=http://192.168.2.94",
    "SEND_RATE=2.0",
    "POLL_RATE=5.0",
    "LOG_NMEA=--log_nmea"
  ]
}
~~~

You should see logs showing up in the BlueOS File Browser, drill into `home > extensions > wl_ugps_external > logs`. They are organized by date.

[image]

clydemcqueen, 2024-12-11:
Mostly done... we have GGA and HDT logs, but no PASHR logs. We forgot to update the sat compass configuration to publish PASHR data.

I think we should do this in a few steps:
1. Start publishing PASHR, do a dive, collect data. Could be off the pier or off the vessel / dock. A Jan/Feb task.
2. After we are happy with the data we can turn off the HDT publishing.

zhrandell, 2025-02-05:
A quick note that one of our winter "to-do" items is to upgrade the GNSS compass firmware (and the G2 box firmware, DVL firmware, etc.), so once we do all that, we could do a quick test flight off Pier 59 to make sure everything is working properly; on that same flight we can also test publish the PASHR data.