Seattle-Aquarium/CCR_development #16: Log the individual beam distances from the DVL (beam splitter)

clydemcqueen, 2024-09-12:
Log all of the distances reported from the DVL, not just the composite.

If we send these as DISTANCE_SENSOR messages, then they should be logged *and* we can use them in SURFTRAK mode.

[See the 1-pager for a detailed description](https://github.com/zhrandell/Seattle_Aquarium_CCR_development/blob/main/1-pagers/DVL_beam_splitter.md)

keenanjohnson, 2025-05-20:
Seems like this should be pretty straightforward. I think DVL sensors are awesome and when I saw this issue I couldn't help look into it a bit. 

The protocol for [the DVL is documented here.](https://docs.waterlinked.com/dvl/dvl-protocol/#ethernet-protocol-tcp)

From the documentation, this is an example of a velocity message, which is parsed by the BlueOS extension. 

```
{
  "time": 106.3935775756836,
  "vx": -3.713480691658333e-05,
  "vy": 5.703703573090024e-05,
  "vz": 2.4990416932269e-05,
  "fom": 0.00016016385052353144,
  "covariance": [
    [
      2.4471841442164077e-08,
      -3.3937477272871774e-09,
      -1.6659699175747278e-09
    ],
    [
      -3.3937477272871774e-09,
      1.4654466085062268e-08,
      4.0409570134514183e-10
    ],
    [
      -1.6659699175747278e-09,
      4.0409570134514183e-10,
      1.5971971523143225e-09
    ]
  ],
  "altitude": 0.4949815273284912,
  "transducers": [
    {
      "id": 0,
      "velocity": 0.00010825289791682735,
      "distance": 0.5568000078201294,
      "rssi": -30.494251251220703,
      "nsd": -88.73271179199219,
      "beam_valid": true
    },
    {
      "id": 1,
      "velocity": -1.4719001228513662e-05,
      "distance": 0.5663999915122986,
      "rssi": -31.095735549926758,
      "nsd": -89.5116958618164,
      "beam_valid": true
    },
    {
      "id": 2,
      "velocity": 2.7863150535267778e-05,
      "distance": 0.537600040435791,
      "rssi": -27.180519104003906,
      "nsd": -96.98075103759766,
      "beam_valid": true
    },
    {
      "id": 3,
      "velocity": 1.9419496311456896e-05,
      "distance": 0.5472000241279602,
      "rssi": -28.006759643554688,
      "nsd": -88.32147216796875,
      "beam_valid": true
    }
  ],
  "velocity_valid": true,
  "status": 0,
  "format": "json_v3.1",
  "type": "velocity",
  "time_of_validity": 1638191471563017,
  "time_of_transmission": 1638191471752336
}
```

The code in the example extension only pulls out the main distance as shown below.

https://github.com/bluerobotics/BlueOS-Water-Linked-DVL/blob/89621f63c560afc6b5919de9008f97b67c9bcb6a/dvl-a50/dvl.py#L359-L405

It will be straightforward to additionally parse out the individual distance measurements from the "transducers" json field, publish them and build a simple GUI extension to display them.

keenanjohnson, 2025-06-17:
Hey all! I think I have the basics of this complete. See my forked extension and PR here: https://github.com/keenanjohnson/BlueOS-Water-Linked-DVL/pull/1

[image]

I haven't published it to the docker hub or anything yet, but can easily if that's the next step and things look good to you all @clydemcqueen and @zhrandell

keenanjohnson, 2025-06-22:
Great! glad I could help.

keenanjohnson, 2025-07-21:
Attaching the log from our testing day! It was a success! 

[00055-2025-07-16_22-16-27.tlog.zip](https://github.com/user-attachments/files/21338668/00055-2025-07-16_22-16-27.tlog.zip)

[image]

keenanjohnson, 2025-07-21:
I have opened a PR to merge the changes here upstream to the official extension. Once that lands, this feature will be available in the official extension and will close out this work. https://github.com/bluerobotics/BlueOS-Water-Linked-DVL/pull/54

keenanjohnson, 2025-07-21:
I have also updated the DVL orientations in this commit, so if the extension is updated before the next ROV flight, the logged orientations of the beams should be correct. https://github.com/keenanjohnson/BlueOS-Water-Linked-DVL/commit/7cce3d4776410863da830be17626815c1150ebc8

zhrandell, 2025-08-20:
Closing this for now as all appears to be working great with the beam splitter. Thanks again, Keenan! Tomorrow we'll get it installed on our other ROV and we'll gather data while conducting real-world ROV surveys.