Seattle-Aquarium/CCR_development #17: open-source image processing and photogrammetry software

zhrandell, 2024-09-15:
We have encountered challenges obtaining the "standard" software for photogrammetry (e.g., [Metashape Agisoft](https://www.agisoft.com/)) at the Aquarium. In keeping with the spirit of our open-source approach, I'd like to consider some open-source image-processing and photogrammetry methods. 

@tonioteran, I outlined some ideas in the [photogrammetry section](https://github.com/zhrandell/Seattle_Aquarium_CCR_development/blob/main/data-pipeline/pipeline-2.md#photogrammetry) of our pipeline-2.md document. I know COLMAP and nerfstudio were the two open-source avenues pursued summer 2023 as part of our MIT SGI work . . . are those still our best approach? What else ought we consider re: image processing (and see below) and photogrammetry?

Additionally, based on the experience implementing those programs on underwater imagery (specifically, where the water itself registered as an object), it may be worth implemented something like [sea-thru](https://openaccess.thecvf.com/content_CVPR_2019/papers/Akkaynak_Sea-Thru_A_Method_for_Removing_Water_From_Underwater_Images_CVPR_2019_paper.pdf), which I see is on GitHub [here](https://github.com/hainh/sea-thru). In fact, sea-thru -- if it works well -- would likely be worth integrating into our standard image processing workflow. @m-h-williams, a test implementation of sea-thru could be a great early-days problem for our new team member to tackle.

clydemcqueen, 2024-10-01:
You may want to break this into 2 issues:
* select an open-source photogrammetry package
* select a color correction tool

zhrandell, 2024-10-01:
Let's have this Issue remain active for photogrammetry methods, and now image processing (color-correction, etc.), can be discussed here - https://github.com/zhrandell/Seattle_Aquarium_CCR_development/issues/21

zhrandell, 2025-01-16:
I'm going to close this one for now. There are several photogrammetry options we will be exploring over the next couple months, but I don't think we need this as an active Issue right now. I'll re-open this down the road, if appropriate.

closed 2025-01-16 by zhrandell