Seattle-Aquarium/CCR_development #18: Downsampling Approach for 1-Hz Telemetry Data

m-h-williams, 2024-09-16:
@zhrandell @clydemcqueen As part of our transition from QGC telemetry files to MAVLink files, we need to downsample telemetry data to 1Hz. There are multiple approaches to downsampling, and we need to decide how we would like to handle this.

Options for downsampling:
1. One Message Per Second: select the first telemetry message received in each second interval. 
2. Take an average: average all messages within a one-second interval.
3. Other suggestions? 

Questions:
What is the preferred downsampling method for our use case?
Are there specific fields where averaging would be inappropriate or inaccurate?

clydemcqueen, 2024-09-16:
Here's how QGC does it, in case you want to emulate that:
* there is a list of "facts", where each fact corresponds to a column in the csv file
* for every telemetry message you see, update the relevant facts
* once per second, write a row with all of the current facts

 [source code](https://github.com/mavlink/qgroundcontrol/blob/606910ddecde1ddcba8347d454fd39a4a5c3bb19/src/Vehicle/Vehicle.cc#L3507)

This is basically method 1, except that it takes the last message of the interval (not the first), and it works even if no message arrived in the interval (the value is just repeated from the previous row).

m-h-williams, 2024-10-15:
The fields (column names) we are currently using in QGC are as follows: 

                  depth=altitudeRelative, 
                  time=clock.currentTime,
                  date=clock.currentDate,
                  lat=gps.lat,
                  lon=gps.lon,
                  DVLx=localPosition.x,
                  DVLy=localPosition.y,
                  altitude=apmSubInfo.rangefinderDistance,
                  temp=temperature.temperature2

I have found the following MAVlink messages correspond to these fields: 
- depth = VFR_HUD.alt
- time = SYSTEM_TIME.time_unix_usec
- date = SYSTEM_TIME.time_unix_usec
- lat = GLOBAL_POSITION_INT.lat
- lon = GLOBAL_POSITION_INT.lon
- DVLx = LOCAL_POSITION_NED.x
- DVLy = LOCAL_POSITION_NED.y
- altitude = RANGEFINDER.distance (cm)

clydemcqueen, 2024-10-17:
I read some code, and these look mostly correct. But there are some surprises:
* gps.lat and gps.lon come from GPS_RAW_INT (from the G2 box), not GLOBAL_POSITION_INT (from the EKF)
* heading comes from ATTITUDE.yaw

The upshot is that the csv file doesn't seem to depend on GLOBAL_POSITION_INT at all. QGC shows the filtered position in the GUI, but it is the raw reading that shows up in the logs. So basically the GUI shows the GPS origin + the DVL track, while the csv file shows the actual GPS track.

The best timestamp might be the timestamp on the MAVLink message in the tlog file. That is the laptop time when QGC received the message.

Happy to chat more.

m-h-williams, 2024-10-17:
Thanks @clydemcqueen!

While QGC uses GPS_RAW_INT for lat/lon, would using GLOBAL_POSITION_INT be a better approximate of the actual ROV location since it uses sensor fusion? Or do you think GPS_RAW_INT would provide more accurate lat/lon readings?

My understanding from the MAVlink common message website was that SYSTEM_TIME is the computer clock. Are they talking about a different computer?
[image]

Thanks again!

clydemcqueen, 2024-10-17:
The EKF is not fusing GPS [yet]. You are getting the EKF (x, y) results already in DVLx and DVLy.

The GLOBAL_POSITION_INT.lat/lon is the same data as DVLx/DVLy, except that it is shifted by the origin.

Until we are able to fuse the GPS data we're stuck with these 2 tracks: DVLx/DVLy is more precise, gps.lat/gps.lon is more accurate over time.

Re: time -- I haven't looked, but I assume that the SYSTEM_TIME value will be the same as the Pi and laptop time. I was thinking that it would be easier to use the timestamp for each message... but either should work.

m-h-williams, 2024-10-18:
Ah I see. Thank you for clarifying, Clyde! I will change adjust the script I'm working on to account for the GPS lat/lon (i.e. GPS_RAW_INT).

m-h-williams, 2024-10-24:
I’ve created a script to generate a 1-Hz telemetry file. Unlike QGC, we opted to average the messages from the .tlog file for downsampling to 1-Hz. You can find it [here].(https://github.com/zhrandell/Seattle_Aquarium_CCR_analytical_resources/blob/main/code/ROV_telemetry_map.py) 

Additional features include:
- User prompts to locate the .tlog file, input the start and end times of the surveys to clip the file, specify the site name, and creating a transects folder to hold the files.
- Calculation of DVLlat and DVLlon based on initial GPS coordinates and DVLx, DVLy movements.
- Area and width of the GoPro-captured images based on altitude.
- A map displaying the survey tracks derived from GPS and DVL data.

Notables:
- I introduced a scale factor to ensure accurate DVLlat and DVLlon coordinates. I derived this from ROV/diver overlap data at Centennial Park, where we knew the transect length. The current scale factor in the script is 0.00227306, calculated by dividing the known survey length (93m) by the distance between the first and last DVL coordinates (40914m). While this scale factor worked for that specific survey, applying it to a survey at Elliott Bay riprap produced overly short and inaccurate DVL tracks. This suggests that different sites may require different scale factors. I’m hoping the EKF will help address this since we won’t always have ground-truth distance data at each site.
- Another note: the reference width and height used to calculate image width and area will need adjustment once we establish a cropping method to reduce GoPro image distortion.

Yay progress!

closed 2024-10-24 by m-h-williams