Seattle-Aquarium/CCR_development #20: Generate the best possible trajectory for each transect (offline)

clydemcqueen, 2024-09-27:
Create an offline (post-process) EKF to generate the best possible trajectory for a transect.

This work may help us understand how to use the ArduSub AHRS to produce better online pose estimates, or it might suggest improvements we can make to the ArduSub AHRS.

Details here: https://github.com/zhrandell/Seattle_Aquarium_CCR_development/blob/main/1-pagers/USBL_DVL_post-processing.md

zhrandell, 2024-11-20:
Tagging Justin's repo where he's working on code for an offline EKF! 

https://github.com/SH0G0TH/CCR-Location-Filtering

zhrandell, 2025-02-05:
Closing this Issue for now as fusion within the ArduSub EKF is taking place using the DVL, UGPS, IMU, and compass data. Specifically, `EKF_SRC1_VELXY` is set to 6, indicating the EKF is using the GPS as an absolute reference, and the DVL as a relative (velocity) reference. 

The black in the attached below is the UGPS, and the red is the fused EKF output

[image]

closed 2025-02-05 by zhrandell