Seattle-Aquarium/CCR_development #24: Python script for survey imagery management

zhrandell, 2024-12-20:
We require a Python script to operationalize various imagery management functions. We want to be able to access it via the command line, similar to our TLOG management script (see, e.g., [here](https://github.com/zhrandell/Seattle_Aquarium_CCR_analytical_resources/blob/main/code/tlog_transects_to_csv.py)).

Desired functions include the following: 

(1)  delete folders of old images. We have redundancy baked into our image cleaning process, and once we are fully happy with cleaned images, some degree of house keeping will be desired. We want to be able to point the script to a folder, resulting in the deletion of the folder and all contents. BONUS - provide the script a list of folders, resulting in a streamlined deletion process. 

(2) copy the contents of one folder into a new folder. We want to name the new folder and specify the file path. For example, if we have imagery stored with the following file path structure: 

```mermaid
---
title: Folder structure for ROV survey imagery
---
graph LR
    CCR[Coastal_Climate_Resilience DropBox]
    flights[flights]
    project[project_ID]
    date[date_and_location]
    camera[camera_angle]
    imagery_type[imagery_type]
    transects[transects]
    transect_ID[transect_ID]
    edited[edited]
    photos[.jpgs]

    CCR --> flights --> project --> date --> camera --> imagery_type --> transects --> transect_ID --> edited --> photos
``` 

with a specific example of folder paths as follows (to [this folder](https://www.dropbox.com/scl/fo/nkgka51g6zmk94c3je1zm/APA28IzNJSZ-_4uRkBHgLk0?rlkey=p7knm31b0la2kudx235fx3h72&st=uyxi797m&dl=0) on DropBox)

```mermaid
---
title: Example folder structure for ROV survey imagery
---
graph LR
    CCR[Coastal_Climate_Resilience DropBox]
    flights[flights]
    project[HSIL]
    date[2024_10_08_diver-ROV_Centennial_Park]
    camera[downward]
    imagery_type[photos]
    transects[transects]
    transect_ID[T4_shallow]
    edited[edited]
    photos[.jpgs]

    CCR --> flights --> project --> date --> camera --> imagery_type --> transects --> transect_ID --> edited --> photos
``` 

we might want to copy a transects worth of imagery (e.g., the .jpgs associated with T4_shallow) to a new folder titled "ML" at the same level of the transects folder. For example, such a ML folder could consist of: 

```mermaid
---
title: Folder structure for all files from a single ROV survey day
---
graph TD
    K[all above folders designating project --> date --> photos]
    K --> A[transects]
    A --> B[T4_shallow]
    A --> C[T1, T2, T3, etc., . . . for all transects within a site]
    B --> D[edited]
    D --> E[original, cleaned .jpgs]
    B --> F[ML]
    F --> G[training]
    F --> H[testing]
    G --> I[.jpgs to train AI on]
    H --> J[.jpgs to apply trained AI on]
```

Furthermore, we will want the ability to execute additional functions when populating the new `training` and `testing` folders. For example, we may want to copy the first half of `edited` into `training`, and the second half to `testing`. Or, we may want to copy every xth image to one of those folders (e.g., every 2nd, or every 3rd image). So, we'll want a couple functions that can be invoked to split up and copy imagery however we require.

m-h-williams, 2024-12-20:
I have created a couple scripts that can be modified for this. One moves images from the RAW image folder to a new folder based on the start and end times of the transect csv file: 

```
# This script can be found in Aquarium Dropbox\Coastal_Climate_Resilience\code\Python 
# File name is move_images_csv_start_end.py 

import pandas as pd
import os
import shutil

# Prompt user for file paths
csv_path = input("Enter the path to your CSV file: ").strip()
image_folder = input("Enter the path to your RAW images folder: ").strip()
destination_folder = input("Enter the path to the destination folder: ").strip()

# Load the CSV file and parse DateTime
df = pd.read_csv(csv_path)
df['DateTime'] = pd.to_datetime(df['Date'] + ' ' + df['Time'], errors='coerce')

# Filter out any rows with invalid dates
df = df.dropna(subset=['DateTime'])

# Automatically get the start and end times from the DateTime column
start_time = df['DateTime'].iloc[0]  # First timestamp in the CSV
end_time = df['DateTime'].iloc[-1]   # Last timestamp in the CSV

# Debugging output: Print the start and end times for reference
print(f"Start Time: {start_time}")
print(f"End Time: {end_time}")

# Filter rows within the specified time range
df_filtered = df[(df['DateTime'] >= start_time) & (df['DateTime'] <= end_time)]

# Ensure the destination folder exists
os.makedirs(destination_folder, exist_ok=True)

# Function to move images within the time range
def move_images_by_time_range(df_range):
    moved_images = 0
    for _, row in df_range.iterrows():
        image_time_str = row['DateTime'].strftime('%Y_%m_%d_%H-%M-%S')

        # Loop through image files in the folder
        for image_name in os.listdir(image_folder):
            if image_name.endswith('.GPR') and image_name.startswith(image_time_str):
                image_path = os.path.join(image_folder, image_name)
                destination_path = os.path.join(destination_folder, image_name)

                # Move the image and confirm
                shutil.move(image_path, destination_path)
                moved_images += 1
                print(f"Moved: {image_name}")
                break
        else:
            print(f"No matching image found for timestamp: {image_time_str}")

    print(f"\nTotal images moved to {destination_folder}: {moved_images}")

# Move images within the time range
move_images_by_time_range(df_filtered)
```

Another keeps one image and removes the next two in a folder, this can be modified to any interval desired: 
```
# This script can be found in Aquarium Dropbox\Coastal_Climate_Resilience\code\Python 
# File name is keep_1_remove_2_images.py 

import os

# Prompt user for the folder containing the .GPR photos
image_folder = input("Enter the path to your GPR images folder: ").strip()

# Get a sorted list of GPR images in the folder
gpr_images = sorted([img for img in os.listdir(image_folder) if img.endswith('.GPR')])

# Counter to track the current image index in the sequence
for i, image_name in enumerate(gpr_images):
    image_path = os.path.join(image_folder, image_name)
    
    # Only keep every third image (1st, 4th, 7th, etc.)
    if i % 3 != 0:
        # Remove the file and confirm
        os.remove(image_path)
        print(f"Removed: {image_name}")
    else:
        print(f"Kept: {image_name}")

print("Operation completed.")
```

reidtosa, 2025-01-07:
```
# File name is file_manager.py

import os
import shutil
from pathlib import Path

# User enters file path to directory
dir_path = input('Enter file path to directory: ').strip()

# Displays user menu
def show_menu():
	print('\nSelect command:')
	print('1. Sort images')
	print('2. Delete directory')
	print('3. Create new directory')
	print('4. Change directory path')
	print('5. Exit')

### Operable Functions ###
# Function to copy images into 'testing' and 'training' folders | *requires nested 'edited' folder within given directory
def sort_images(folder):
	# Locates the 'edited' folder within the transect folders
	edited_dir_path = Path(folder) / "edited"
	if not edited_dir_path.exists():
		print(f"Error: No 'edited' folder found in {dir_path}")
		return

	# Creates a reference list of all images in the 'edited' folder
	image_list = [file for file in os.listdir(edited_dir_path)
		if file.lower().endswith('.jpg') and os.path.isfile(os.path.join(edited_dir_path, file))]

	# Creates 'testing' and 'training' folders
	testing_path = os.path.join(dir_path, 'testing')
	os.makedirs(testing_path, exist_ok = True)
	training_path = os.path.join(dir_path, 'training')
	os.makedirs(training_path, exist_ok = True)

	# Copies every third image to the 'testing' folder and all others to the 'training' folder
	for i, image in enumerate(image_list):
			edited = os.path.join(edited_dir_path, image)
			if i % 3 == 0:
				testing_folder = os.path.join(testing_path, image)
				shutil.copy(edited, testing_folder)
				print(f"Copied {image} to testing")
			else:
				training_folder = os.path.join(training_path, image)
				shutil.copy(edited, training_folder)
				print(f"Copied {image} to training")

# Function to delete file from file path
def delete_file(folder):
	if os.path.exists(folder):
		os.chmod(folder, 0o777) # changes folder permission to full access
	shutil.rmtree(folder)
	print(f"Successfully deleted {os.path.basename(folder)}")

# Function to create a new directory
def create_directory(name, path):
	new_dir_path = os.path.join(path, name)
	if not os.path.exists(new_dir_path):
		os.mkdir(new_dir_path)
		print(f"Successfully created directory '{name}' under '{path}'")
	else:
		print(f"Error: Directory '{name}' already exists at '{path}'")

# Function to change current working directory
def change_directory():
	delta_dir_path = input('Enter new directory path: ').strip()
	if os.path.isdir(delta_dir_path):
		return delta_dir_path
	else:
		print(f"Error: The directory '{delta_dir_path}' does not exist.")
		return None

### Execution of Command ###
# Operates user menu
def main():
	menu()

def menu():
	global dir_path
	while True:
		show_menu()
		try:
			command = int(input('\nInput: '))
		except ValueError:
			print("Invalid input, please enter a valid number.")
			continue
		# Calls function to sort images into generated 'testing' and 'training' folders
		if command == 1:
			sort_images(dir_path)
		# Calls function to delete a directory and its contents
		elif command == 2:
			print('Enter paths to directories that will be deleted')
			delete_this = input('Separate multiple paths with spaces: ').strip()
			print(f"\n*WARNING: File deletion is irreversible*\n'{os.path.basename(delete_this)}' and its contents will be permanently deleted\n")
			while True:
				confirm = input('Proceed with directory deletion (Y/N)?: ').strip()
				if confirm.lower() == 'y':
					delete_file(delete_this)
					break
				elif confirm.lower() == 'n':
					print('\nOperation canceled')
					break
				else:
					print("Invalid input, please enter 'y' or 'n'")
		# Calls function to create a new directory	
		elif command == 3:
			name_dir = input('Enter name for new directory: ').strip()
			create_directory(name_dir, dir_path)
		# Change directory
		elif command == 4:
			new_dir_path = change_directory()
			if new_dir_path:
				dir_path = new_dir_path
		# Terminate script
		elif command == 5:
			print("Exiting the program.")
			break
		else:
			print(f'\n{command} is invalid, please select a valid command.')
menu()
```

@zhrandell here is what I've come up with so far. I'll need a bit more time to implement multiple inputs for deletion and any other features, but this version is ready for use!

reidtosa, 2025-01-15:
file_manager.py allows the user to sort image shortcuts from transects, delete folders with images, and create new folders. It is located under the following path: 
"Seattle Aquarium Dropbox\Coastal_Climate_Resilience\GitHub\Seattle_Aquarium_CCR_analytical_resources\code\file_manager.py"

The version commented above was updated as of 2025-01-15 and will continue to receive small adjustments.

closed 2025-01-15 by reidtosa