Seattle-Aquarium/CCR_development #29: AI/ML to help process GoPro .GPR photos

zhrandell, 2025-04-12:
Creating an Issue for easy communication re: our newest 1-pager, [AI/ML_image_processing](https://github.com/Seattle-Aquarium/CCR_development/blob/main/1-pagers/AI-ML_image_processing.md)

It appears there are a couple of options available. At the moment, we're aware of a split between [_**1**_] off-the-shelf software, e.g., [Aftershoot](https://aftershoot.com/edit/) and [Imagen](https://imagen-ai.com/) and [_**2**_] building our own AI/ML photo editing workflow in Python. 

Regardless of approach, we'd ideally like (need?) to interface with Adobe Lightroom Classic on some level, as the AI/ML processed photos at minimum will require review, and likely may still require tuning. Off-the-shelf programs could likely plug in directly (as it appears Imagen linked above does) though I'm uncertain the degree to which Python scripts can interface with Adobe Lightroom Classic. 

Potential next steps include: 
- [ ] explore Aftershoot in greater depth
- [ ] explore Imagen in greater depth
- [ ] explore custom Python options
- [ ] explore the degree to which Python can interface with Adobe Lightroom Classic

zhrandell, 2025-04-12:
re: the fourth bullet-point above, it doesn't look like Adobe Lightroom Classic has any sort of Python API, but it does have a software development kit (SDK), specifically [Adobe Lightroom SDK](https://developer.adobe.com/lightroom-classic/). This enables Lua scripts to be invoked to enable _some_ degree of automation. However, it looks like more advanced features are not accessible via Lua scripting. 

It appears however that we might be able to invoke a combination of approaches. For example, you can review a [ChatGPT query here](https://chatgpt.com/share/67fa84b2-66fc-8000-9f70-3c83abbb5965) that elaborates on what is possible with Lua scripting and the Adobe SDK in combination with Python ahead of time.

zhrandell, 2025-04-12:
one consideration re: using Python for an AI/ML photo editing workflow is that it appears certain libraries require the .GPR files to be converted to .TIFF. This would (likely) be undesireable, as doing so would reduce our customization and flexbility (it looks like we'd lose the sensor-level flexibility associated with .GPR photos). 

After digging in further, it looks like [rawpy](https://pypi.org/project/rawpy/) is a library that will allow us to work with .GPR files, as GPRs are a derivation of .DNG (Digital Negative). These raw files store Bayer-pattern data, meaning pixels are not yet interpolated. rawpy would allow us to input .GPR files as Bayer-pattern arrays, can demosiac and output RGB images to NumPy arrays. 

One limitation however is that as .GPR is a proprietary format, it doesn't look like rawpy (or other libraries) can write .GPR files. This means that rawpy (or other Python AI/ML libraries) would need to export something like a .TIFF. Depending upon the efficacy of a AI/ML pipeline, this may be (or may not) be sufficient.  

So, one possible workflow could look like the following: 

```mermaid
---
title: potential AI/ML assisted workflow to process photos
---
graph LR
    A[.GPR raw photo]
    B[rawpy -> RGB NumPy array]
    C[AI/ML via PyTorch, TensorFlow, etc.]
    D[export .TIFF]
    E[possible Lua scripting via Lightroom SDK]
    F[final review/processing of .TIFF in Lightroom]
    G[export final .JPEG from Lightroom]
    A --> B --> C --> D --> E --> F --> G
```

zhrandell, 2025-07-17:
Updating this Issue to note that we have set up a [Google TPU Research Cloud](https://sites.research.google/trc/about/) project to tackle this. The intent is to explore using GANs (Generative Adversarial Networks) to train on our over 5000 sets of (1) pre-processed (raw) imagery and (2) processed imagery. 

The ultimate objective is to have a trained model that can be applied to unseen imagery such that we are able to process it, thus removing the biggest rate-limiting step for our research.

keenanjohnson, 2025-07-18:
@zhrandell  you mentioned that one solution limitation might be that the .gpr format is proprietary. Although that's true there is a great open source library that is published that allows you to convert images into or out of the .gpr format. I've linked it below. 

If I am understanding the workflow above, that would allow us to break away from the dependency of using lightroom which is probably a win given the challenges of licensing etc? 

https://github.com/gopro/gpr

zhrandell, 2025-07-19:
> [@zhrandell](https://github.com/zhrandell) you mentioned that one solution limitation might be that the .gpr format is proprietary. Although that's true there is a great open source library that is [...]

Nice, thanks @keenanjohnson! Tagging @tonioteran to make sure he's aware of that repo for working with raw .GPR files. Yes, we definitely want to remove Adobe Lightroom from the equation as much as possible (perhaps, if needed, Adobe can be used at the very end to put any final touches on the images? but the more we can do via an automated pipeline, the better!). 

Tonio started to configure a Google Cloud project such that we have a VM with PyTorch and Cuda cued up . . . I believe he is double-checking what precise image processing methods to try to replicate in that context. 

Two pieces still needed on our end are to (1) set up the appropriate file structure in the cloud, and (2) populate with pre- and post-processed image pairs into the cloud . . . Tonio indicated that we can easily do the latter via a scp; it looks to be relatively simple, but I haven't done that before, so I'll need to figure that out.

reidtosa, 2026-09-10:
Closing with the latest:

**LUT Model**
@keenanjohnson found and trialed the 3D Lookup Tables (LUT) model with much success (see repo: https://github.com/HuiZeng/Image-Adaptive-3DLUT). Additional tests on new images yielded promising results, albeit with some kinks that may require a closer look (namely, persistent green splotches in some images that have popped up across model types; see below).

<p align="center">
  [image]
</p>
<p align="center">
  <em>Given the silty bottom at our Fish Pens site, we would expect a homogenous neutral gray seafloor.</em>
</p>

**Automated .GPR Pre-processing**
With much Clauding, we now have the means to automatically pre-process (crop, remove chromatic aberration, denoise, and export) our .GPR files into .TIFF files that may then be fed into the [UIE](https://github.com/Seattle-Aquarium/underwater-auto-image-encoder). Our team is [developing a GUI](https://github.com/Seattle-Aquarium/CCR_ROV_survey_methods/tree/main/UTC) that ties together many of the standard scripts and processes applied to data collected during and after flight.

Further developments on the topic of automated image processing may be followed in [CCR_image_processing](https://github.com/Seattle-Aquarium/CCR_image_processing).

closed 2026-09-10 by reidtosa