Seattle-Aquarium/CCR_development #31: Improve SURFTRAK control

clydemcqueen, 2025-05-14:
We'd like to improve the SURFTRAK pilot experience before the summer 2025 surveys.

Operating principle:
* If the pilot has SURFTRAK engaged they want a specific target -- usually 0.8m. The pilot may need to move the ROV up/down or left/right quickly, but if SURFTRAK is still engaged they want to get back to the desired target as soon as possible.

Problems with the current system (ArduSub + Lua script + QGC + Joystick mappings):
* The current rangefinder target is buried in the messages box in QGC, so it takes a few clicks to see the current target. It is also available in surftrak_fixit, but this takes valuable screen real estate, or requires clicks.
* It is too easy to accidentally nudge the up/down on the right joystick, changing the target. Getting back to the desired target takes a few joystick actions with the current Lua script, in part because the current script won't allow a target < 1m.
* If there is a significant transverse current -- requiring semi-constant strafing thrust -- SURFTRAK is basically unusable.
* Semi-related: can we use the trigger controls as buttons? This would give us more control options.

Proposed tasks / solutions:
- [ ] Confirm that Cockpit will display the rangefinder target
- [ ] See if Cockpit will map trigger controls to buttons
- [ ] Allow targets as low as 0.5m (Lua)
- [ ] Keep track of the desired rangefinder target, and re-set the target if SURFTRAK changes it (Lua)
- [ ] Set up a SITL test with the bench Pi, new script(s), Cockpit

@m-h-williams Thoughts?

clydemcqueen, 2025-05-14:
Cockpit will allow you to map the trigger controls to various actions. 2 more buttons!

clydemcqueen, 2025-05-15:
Yes, Cockpit will show RF Target. 

[image]

clydemcqueen, 2025-05-15:
Cockpit displays RF Target. 

[image]

m-h-williams, 2025-05-15:
@clydemcqueen Looks great! Thanks for getting this rolling Clyde. I'm excited to use cockpit! It seems to have the helpful customization capabilities. 

Other than downloading the extension on BlueOS, do you know if there are any other steps that will need to take place before we can switch over from QGC to Cockpit? I can dig into this, but thought I would ask in case you have already done the homework.

clydemcqueen, 2025-05-16:
@m-h-williams Cockpit is crazy customizable. Docs here: https://blueos.cloud/cockpit/docs/latest/usage/overview/

We can use the bench Pi to set up a simulation that we can use to play with the new surftrak script & also experiment with cockpit. I can drop it off next week.

m-h-williams, 2025-05-19:
@clydemcqueen sounds like a great idea. No huge rush at the moment, if you are able to drop it off by early next week that would be great!

clydemcqueen, 2025-05-22:
Closing this for now... lmk if something comes up!

closed 2025-05-22 by clydemcqueen