Seattle-Aquarium/CCR_development #33: Will beam splitter improve surftrak?

clydemcqueen, 2025-07-17:
Analyze the tlog files generated by beam splitter (16-July-2025 test, etc.) and evaluate whether or not beam splitter data can improve surftrak behavior.

clydemcqueen, 2025-11-20:
Summarizing our discussion from today:
* I looked at `00118-2025-10-08_18-03-06.tlog`
* There were 94401 DISTANCE_SENSOR messages, but for a bunch I wasn't able to reconstruct the set of 5 messages for a single DVL packet, or something else went wrong (e.g., the ROV was tilted too much). In the end I was able to use 96% of the messages to calculate the terrain slope.
* I am experimenting with a 3-state terrain EKF (terrainState, slope_north, slope_east) to see if we can come up with a reliable estimate for terrainState at T=now. If it works, then I think we can move forward with adding a terrain EKF to surftrak.

I'll post my code soon.

The anomaly we saw was at timestamp 1759948105.802445, which translates to Wednesday, October 8, 2025 11:28:25.802 AM GMT-07:00 DST. @zhrandell

zhrandell, 2025-11-21:
Sounds like a pretty good (and quite plausible) story to me! 

And nice bug squashing . . . that's quite an improvement!