Seattle-Aquarium/CCR_development #34: Re-light Lutris

clydemcqueen, 2025-07-17:
_The problem_

Lutris (our primary research ROV) can take 4 high-lumen dive lights to provide lighting during surveys. Each light requires a battery, which should be changed for every transect. This lighting system slows down operations and adds unnecessary weight and bulk to the vehicle.

_The proposed solution_

Install 4 semi-permanent high-lumen subsea lights on Lutris. Each light should draw power from the existing Outland power system. Provide a way to control these lights remotely from the surface.

zhrandell, 2025-12-12:
Thank you for helping us with this, @keenanjohnson!

zhrandell, 2025-12-15:
This looks great, Keenan - thank you! I don't have the subject matter expertise to evaluate the wire diagram. Re: the housing and the ATO, it sounds like you're confident there will be enough room for everything? I don't have the intuition to guide how much space the wires require, etc., but based on the below image from the BR webpage, we can have everything snugged in rather cozy. Let me know when you're ready for parts to be oredered and we'll make it happen! 

[image]

keenanjohnson, 2025-12-16:
Sounds good. I'll finalize selection of the smaller components today. @clydemcqueen are you aligned with the selection of one of the raspberry pi microcontrollers in the diagram above to do the control? We would need to write some software to interface with ardusub and then the lights, but I think that is worth it for the reduction and cabling complexity.

keenanjohnson, 2025-12-18:
Yes @m-h-williams ! I wasn't too specific as we can use a lot of things. We just need something to connect the single positive and ground wires coming out of the converter to each of the 4 power and grounds of the LEDs. These snap connectors you linked are awesome and as long as the wire gauge we have will fit in them, they would work well!

m-h-williams, 2025-12-18:
Great! Then robotics santa has finished purchasing everything on the list

keenanjohnson, 2025-12-18:
Thank you! Perhaps sometime after the holidays we can have a bench day of wiring everything up and doing our first bench tests with just the LEDs / converter before installing it all on Lutris as @clydemcqueen suggested originally in this post! 

[image]

zhrandell, 2026-01-06:
Yay, thanks @keenanjohnson! And just a quick note that we've received all the hardware components that we ordered! We should be ready to bench test whenever you are.

keenanjohnson, 2026-01-15:
We had a good day of initial wiring and testing of the new lights system!

We successfully tested that:

1. The voltage converted did function as intended!
2. The micropython script ran on the microcontroller (raspberry pi pico) 
3. All the hardware fit well inside the current enclosure (if we choose to stick with this one)
4. One light at max power drew around 98 Watts

We hit two small hicups:

1. I had forgotten to add the RS-485 transceiver to the BOM. That's been resolved and added here: https://github.com/Seattle-Aquarium/CCR_ROV_survey_methods/issues/32

2. The WAGO connector blocks we were using were too small for the input side wires of the current monitor. I'll try to find a few larger ones that we can add to the BOM and use for the input connection

[image]

[image]

[image]

clydemcqueen, 2026-04-03:
* Edit1: Yay, good dive! Big milestone.
* Edit2: Wow, look at Lutris rocking the new look!

I was surprised to see the video show the flickering getting _brighter_. Were the lights mostly at the target illumination, and then they just randomly got brighter, or were you trying to get to that higher illumination level, and they just couldn't get there, except occasionally?

It would be helpful to see the PWM signals for the lights and the thrusters, so we can try to correlate the power fluctuations with the PWM signals, or something else. Is the bin file uploaded somewhere?

And we need to figure out how to get the tlog files working....

zhrandell, 2026-04-03:
Yeah! I believe the youngsters call it a "glow up" - Lutris is now rocking the red trim that matches the Seattle Aquarium diver's drysuit, with red up top and black on the lower 2/3rd. The red was also selected due to providing a nice contrast against the dark sea surface when the vehicle is at the surface.  

Regarding flashing _brighter_, Clyde, I think your observation is correct . . . I think the lights are mostly at the correct output, but then start to flicker at brighter (and darker) levels (but again, I'm not 100% certain . . . ). 

The full bin file for the entire flight can be found [here on DropBox](https://www.dropbox.com/scl/fi/b5x63q81oe2u509s0r4ez/00000090.BIN?rlkey=i3043tfb23ozlj3ultfotmmyk&dl=0) - you should be able to download it. 

Also, note that between `10:36:40` and `10:40:19` (via `TC 25` from the GoPro Labs [time sync page](https://gopro.github.io/labs/control/precisiontime/)) Reid ran a "transect" where the lights were at 50%, and he flew forward at 0.10-0.125m/s. We originally wanted to capture power consumption metrics for this "transect" with the lights at 100%, but we backed down to 50% given the flickering behavior. 

This also leads to one caveat about the flight . . . I was a little alarmed by the flickering, so the vast majority of the time we were down there I signaled to Reid to have the lights at 50% . . . so, there will only be a few select instances within the bin log that capture the "flickering". Does the bin log capture PWM signaling? If so, that'd be very helpful to isolate thruster and power dynamics for any and all instances where the PWM was above 50%.

zhrandell, 2026-04-09:
Thank you, @amarburg and @keenanjohnson! Based on briefly talking to Aaron about this on-site yesterday, it sounds like we indeed simply need to plug into the pico and monitor the PWM signals via a serial port that should be accessible. Aaron also noted that we can up the Hz via modifying a single line of code and pushing those changes to the pico, which may provide better resolution re: the potential occurence of spurious PWM signaling. 

We can definitely plug into the pico, though I'm not precisely sure how to pull up the serial port, and I haven't flashed code to the board either. But, these do seem like very basic things to do, and we should know how to do rudimentary de-bugging on the system like this, so next week we will tackle both of these items. It sounds like the start up guides likely contain this info. I'll ping folks back here if we're stuck and/or what we find. 

Thanks again!

zhrandell, 2026-04-24:
Okay folks, we implemented @keenanjohnson's latest software edits, tested once more (no flickering observed on the bench!), sealed up the box, pressure tested, and flew the ROV off the side of the pier for a real-world test. The results were . . . a little odd. 

The flickering has definitely improved from how bad it was the last time the ROV was underwater (see the video in the [above post](https://github.com/Seattle-Aquarium/CCR_development/issues/34#issuecomment-4181020923) if you want a refresh on what we previously observed). 

We recorded today's full flight with a downward-facing GoPro, and you can view it all [here on DropBox](https://www.dropbox.com/scl/fi/el4f7mpeqjoeik0wajaq2/GX012069.MP4?rlkey=4k9qsd8vmb2aeelbs3bsgvc6j&dl=0). Specifically, if you skip to `7:40`, you'll see the flickering from today at its "worst". Definitely an improvement, but definitely still flickering. 

To deepen the mystery, upon pulling the ROV out of the water, and prior to powering off, we once more powered the lights to 100% and ran the thrusters . . . no flickering (just as when we tested on the bench prior to the flight . . . [see for yourself here](https://www.dropbox.com/scl/fi/jnumg5ryut4fr4ljzwivu/IMG_1656.MOV?rlkey=m005pahz7rg03431kl3lyq2h6&dl=0)). 

The good news is that Keenan's updates have improved the flickering . . . the bad news is that we can no longer reproduce the flickering on the bench, but it still manifests at depth when flying.  

Aside from the flickering, it was a really odd flight . . . none of the flight control modes were working properly (stabilize, depth hold, surftrak) . . . even position hold was acting up, specifically where, in pos hold, the ROV would lurch forward and downward. The full BIN file from that flight is [here on DropBox](https://www.dropbox.com/scl/fi/ch3ap0ibi3ntdtefqyxyi/00000101.BIN?rlkey=trwrilrchwk58euh3gn166gap&dl=0).

amarburg, 2026-05-05:
@zhrandell @clydemcqueen and I spent a couple of hours with Lutris yesterday, this was all testing on the bench with the light enclosure and the main ROV housing opened.  

For testing I reverted the Pico2 to commit [33c90b](33c90b90963a83bdca5106dc2ea5d5608df2ee1d) before @keenanjohnson's recent filtering changes as the old version was known to be _more_ sensitive to the flickering, with `STATUS_INTERVAL_MS = 100` for increased verbosity

We attached the scope the PWM input to the Pico board.   I didn't do a great job of fully characterizing the noise, but here's a sample scope trace.

[image]

In this state, it was easy to reproduce the flickering by turning the lights up and "blipping" the thrusters.

# Attempt 1:  External pulldown resistor

I was fixated on the idea that the [RP2350 pull-down errata](https://hackaday.com/2024/09/20/raspberry-pi-rp2350-e9-erratum-redefined-as-input-mode-leakage-current/) was part of the problem.   We weren't seeing latch-down per se but it could have impacted the PWM measurement.

We installed an external 10kohm pull-down on the Pico board and [disabled the pulldown here](https://github.com/Seattle-Aquarium/CCR_ROV_survey_methods/blob/cad054f96a24422fffcdf4ac041fda8846bbfd8b/lighting/code/src/main.py#L40).   This did not seem to affect the behavior.  Removed the external resistor and re-enabled the pulldown in code.

# Attempt 2:  Twisted pair

Using Zach's ping cable above, we attached the signal (green) wire at the navigator board, and both the signal and ground (black) wires at the Pico end, to see if the twisted pair would act as a shield.     We did not terminate ground on the Navigator end as this would have created a ground loop.

Lights still flickered under the configuration.  Might have flickered slightly less?

# Attempt 3:  

Finally, we tried a more radical solution to isolate the PWM signal and Pico from the light and motor power bus.   We connected all three lines from the Navigator output (5V, PWM @ 3v3, GND) through Zach's Ping cable, and used that 5V supply and ground to power the Pico board and RS485 converter.   

[image]

To do this without a ground loop, we also:

* Completely disconnected the 24V input to the pico board
* Removed the GND from the RS485 output.

This solution relies on RS485's common mode rejection to manage any ground differences between the Pico and the lights.

This seemed to work on the bench; we did not observe any flickering.

We weren't able to install the new Ping cable, so TBD whether having the cable fully installed changes behavior.

At the end of testing I also restored Pico firmware to the [current commit](cad054f96a24422fffcdf4ac041fda8846bbfd8b) with @keenanjohnson's robustness changes.

Sounds like @zhrandell will be able to test the ROV early next week.

amarburg, 2026-09-30:
Brief update.   I have run USB from the Pi to the Pico in the light controller.   The existing PWM signal is also passed through.     

Lutris is buttoned up on the bench with the current software (see [CCR_ROV_Survey_methods!43](https://github.com/Seattle-Aquarium/CCR_ROV_survey_methods/pull/43) and [!47](https://github.com/Seattle-Aquarium/CCR_ROV_survey_methods/pull/47).

To support this I installed `mpremote` on the Pi, as well as a git checkout of the CCR_ROV_Survey_methods repo.

I haven't updated the code to provide any feedback from the lights, but now these changes can be done w/o opening the light housing.   Per [!47](https://github.com/Seattle-Aquarium/CCR_ROV_survey_methods/pull/47), I swapped a QtPy2040 in for testing due to some cabling issues but should be trivial to swap back to Keenan's PCB when available -- I will need to come up with a Micro-USB adaptor.

I also need to complete a documentation package (primarily a wire diagram for the new harness).

Next step is in-water testing.