Seattle-Aquarium/CCR_development #35: Move forward at constant velocity while running SURFTRAK

clydemcqueen, 2025-07-17:
_The problem_

We run transects at a constant forward velocity and a constant distance above the seafloor. The distance is maintained by SURFTRAK, but the pilot needs to carefully manage the forward velocity. This increases operational complexity.

Notes:

- the pilot needs to be in control of yaw and strafe during these operations to move around stipes and react to currents
- we can assume the ROV has a DVL and precise local positioning

_The proposed solution_

Add a control scheme that will move the vehicle forward at a constant velocity (e.g., 10cm/s) while running SURFTRAK.

Components:

- [ ] modify ArduSub to support the ABOVE_TERRAIN frame during GUIDED mode
- [ ] expose this to Lua scripts
- [ ] write a Lua script that will use this to move forward at the current yaw at a constant velocity

keenanjohnson, 2025-12-09:
You rock @clydemcqueen ! I know that wrangling ardusub dependancies is a real challenge sometimes!

clydemcqueen, 2026-01-08:
We had a successful test off Pier 59 yesterday. The logs look pretty good. Yay!

We left the custom build on Nereo for now, along with a Lua script _transect3.lua_.

A few nits:
* The Lua script should increment / decrement the rangefinder target by something smaller than 10cm. I'll make this a parameter with a default of 5cm.
* Same for WPNAV_SPEED (5 cm/s increment)
* Ideally a speed change would be reflected immediately. Maybe I can do this in Lua... TBD
* BlueOS didn't show the "script1" tags in the parameter change UI. Might be a BlueOS bug?
* We noticed that the battery voltage values were weird in Cockpit. Maybe a config error?

More later...

clydemcqueen, 2026-09-30:
It has been a while since I've provided an update, and there have been a lot of changes!

ArduSub `Guided_PosVel` mode has become `Guided_PosVelAccel` mode, and it is now visible in Lua scripts. These changes were proposed in https://github.com/ArduPilot/ardupilot/pull/33609 and merged into 4.8-DEV a month ago. They were backported and released today in ArduSub-4.7.0-beta1.

The [transect4.lua](https://github.com/clydemcqueen/guided_above_terrain/blob/main/lua/transect4.lua) script was completely rewritten to use this new `Guided_PosVelAccel` mode. `transect4.lua` generates a kinematic trajectory in XYZ to move forward at a constant speed _while_ following the seafloor. We should see a significant improvement in the ability to maintain course and speed in challenging environments. (In simulation it performs well with a 3.5 knot cross-track current.)

Like the older scripts, the new `transect4.lua` script implements a "sticky" rangefinder target that applies across all modes. Pilots can switch modes as necessary during a transect without having to re-set the target. (Disarm to forget the sticky target.)

Other improvements:
* Speed changes take effect immediately.
* The forward motion is throttled if the terrain comes up too fast.
* There are custom parameters to control the behavior, so we can make changes in the field.

Custom parameters:
* `GAT_RF_INC`: sets the rangefinder target increment (default 0.05 m)
* `GAT_SPD`: sets the forward speed during GUIDED mode (default 0.2 m/s)
* `GAT_SPD_INC`: sets the speed increment (default 0.05 m/s)
* `GAT_CLR_SLOW`: sets the floor clearance slowdown threshold in meters (default 0.65 m)
* `GAT_CLR_MIN`: sets the hard-stop minimum floor clearance in meters (default 0.40 m)

Feedback welcome!

zhrandell, 2026-09-30:
Wow this is awesome, congratulations @clydemcqueen for leading the charge and getting this into ArduSub 4.7.0-beta1! 

This is awesome new functionality . . . now that we've finished surveys for the season, I look forward to testing this when we have a chance (and after we've debugged our DVL a bit further . . . ). We're going to need to map some new shift-button functionality on our controller!