Seattle-Aquarium/CCR_development #36: EKF fusion issues with DVL + GPS

m-h-williams, 2025-09-16:
# EKF fusion issues with DVL + GPS

Hi @clydemcqueen, we're running into persistent EKF fusion problems with our DVL + GPS setup and could really use your help troubleshooting.  

## Background  
Fusion issues have persisted since our first survey in Elliott Bay (8/13/2025). Both GPS and DVL appear to be working when checked via their extensions, but when uploading the tlog to UAV Log Viewer no geo-referenced ROV path is visible. The `tlog` also shows no `GLOBAL_POSITION_INT` values, confirming that fusion isn’t taking place.  

## Parameter checks  
Using your `tlog_param.py` script, I confirmed the parameters didn’t match those you outlined [in your Teams post](https://teams.microsoft.com/l/message/19:a0b9a8921ce740a8b42e34c1ea1a04b2@thread.tacv2/1738006036157?tenantId=4fd10c42-3378-4cdb-81b6-123e971b4aad&groupId=28e89234-aab4-4a84-b5ff-9bfadaa4e0c7&parentMessageId=1738006036157&teamName=ROV%20%26%20AI%20methods%20and%20analyses&channelName=Data%20analysis&createdTime=1738006036157).  

Initial values were:  
```
EK3_SRC1_POSXY	6   # ExternalNav
EK3_SRC1_POSZ	1   #Baro  
EK3_SRC1_VELXY    6   #ExternalNav
EK3_SRC1_VELZ	0   #None
EK3_SRC1_YAW	1   #Compass
```

On 8/21/2025, I updated `EK3_SRC1_POSXY` from `6` → `3` :
```
EK3_SRC1_POSXY	3   # ExternalNav
EK3_SRC1_POSZ	1   #Baro   
EK3_SRC1_VELXY    6   #ExternalNav
EK3_SRC1_VELZ	0   #None
EK3_SRC1_YAW	1   #Compass
```

But even after this change, no fusion occurred.  

## Diagnostics  
`tlog_info.py` shows sensor health problems:  
```
0x200000 AHRS subsystem health: enabled 97.53%, healthy NEVER!
0x20 GPS: enabled always, healthy NEVER!
```
Full results here
```
tlog_info.py

Results for 00039-2025-08-21_16-37-51.tlog
    System = Vehicle
        Component = mav_comp_id_autopilot1
            HEARTBEAT
                         551 ALT_HOLD
                        2374 DISARMED
                         115 MANUAL
                         867 POS_HOLD
                        1154 RNG_HOLD
                         141 STABILIZE
                        5202 Total heartbeat messages, approx 86 minutes
            STATUSTEXT
                severities
                           3 CRITICAL
                          15 ERROR
                           6 WARNING
                         134 INFO
                strings
                          14 ArduSub V4.5.3 (80e8c23a) REBOOT!
                          14 805aa88d1a9a498291a0a2d10cabe5df
                          14 Frame: VECTORED_6DOF
                           1 Barometer 1 calibration complete
                           1 Barometer 2 calibration complete
                           1 ArduPilot Ready
                           1 surftrak2.lua running
                           1 AHRS: DCM active
                           1 GPS 1: specified as MAV
                           3 Lost manual control
                          17 EKF3 waiting for GPS config data
                           1 No ap_message for mavlink id (159)
                           1 No ap_message for mavlink id (251)
                           1 No ap_message for mavlink id (134)
                           1 No ap_message for mavlink id (295)
                           1 No ap_message for mavlink id (11030)
                           1 No ap_message for mavlink id (11031)
                           1 No ap_message for mavlink id (11032)
                          13 IMU0: fast sampling enabled 8.0kHz/1.0kHz
                           3 Compass performance degraded
                           1 EKF3 IMU0 initialised
                           1 AHRS: EKF3 active
                           1 EKF3 IMU0 tilt alignment complete
                           1 EKF3 IMU0 MAG0 initial yaw alignment complete
                           2 EKF3 IMU0 fusing odometry
                           1 EKF3 IMU0 switching to compass 1
                           2 EKF3 IMU0 started relative aiding
                           3 #Gain is 40%
                           1 #Gain is 50%
                           1 EKF3 IMU0 stopped aiding
                           1 EKF3 IMU0 MAG1 in-flight yaw alignment complete
                           1 rangefinder target is 0.54 meters
                           2 rangefinder target is 1.00 meters
                           2 rangefinder target is 0.90 meters
                          29 rangefinder target is 0.80 meters
                           2 #Gain is 30%
                           1 #Gain is 20%
                           1 rangefinder target is 0.87 meters
                           1 rangefinder target is 0.65 meters
                           2 rangefinder target is 0.76 meters
                           1 rangefinder target is 0.73 meters
                           1 rangefinder target is 0.75 meters
                           2 rangefinder target is 0.79 meters
                           3 rangefinder target is 0.77 meters
                           2 rangefinder target is 0.78 meters
                           1 rangefinder target is 0.81 meters
                           1 EKF3 IMU0 MAG1 ground mag anomaly, yaw re-aligned
            SYSTEM_TIME
                         valid time was sent 15128 time(s)
            SYS_STATUS
                sensors
                         0x01 3D gyro: enabled always, healthy always
                         0x02 3D accelerometer: enabled always, healthy always
                         0x04 3D magnetometer: enabled always, healthy always
                         0x08 absolute pressure: enabled always, healthy always
                         0x80 computer vision position: enabled always, healthy 88.76%
                         0x100 laser based position (down-facing sonar rangefinder): enabled always, healthy 90.96%
                         0x400 3D angular rate control: enabled always, healthy always
                         0x800 attitude stabilization: enabled always, healthy always
                         0x1000 yaw position: enabled always, healthy always
                         0x2000 z/altitude control: enabled 66.76%, healthy 66.76%
                         0x4000 x/y position control: enabled 66.76%, healthy 66.76%
                         0x8000 motor outputs / control: enabled always, healthy always
                         0x100000 geofence: enabled NEVER!, healthy always
                         0x200000 AHRS subsystem health: enabled 97.53%, healthy NEVER!
                         0x1000000 Logging: enabled always, healthy always
                         0x2000000 Battery: enabled always, healthy always
                         0x10000000 pre-arm check status. Always healthy when armed: enabled always, healthy 99.99%
                         0x20 GPS: enabled always, healthy NEVER!
        Component = mav_comp_id_camera
            HEARTBEAT
                          68 Total heartbeat messages, approx 1 minutes
        Component = mav_comp_id_onboard_computer4
            HEARTBEAT
                        5183 Total heartbeat messages, approx 86 minutes
    System = QGroundControl or similar
        Component = mav_comp_id_missionplanner
            HEARTBEAT
                           7 Total heartbeat messages, approx 0 minutes
            SYSTEM_TIME
                         valid time was sent 2 time(s)
        Component = mav_comp_id_udp_bridge
            HEARTBEAT
                        5215 Total heartbeat messages, approx 86 minutes
```
</details>  

## Additional steps tried  
- Followed Waterlinked’s [DVL + GPS integration docs](https://docs.waterlinked.com/dvl/bluerov-integration-dvl-ugps/) I selected the `LOAD PARAMETERS FOR DVL+GPS` button in the DVL extension on BlueOS. Fusion still failed.  

## Possible next step?
- According to Waterlinked docs, `GPS_TYPE` should be "not set" but it is `14` in the current parameters, and I couldn’t find `EK3_GPS_TYPE` in the file.  

[image]  

---

**Question:** Do you see anything obvious in my param setup or sensor health output that explains why EKF isn’t fusing GPS + DVL?  

I feel like I have reached a dead end, so your expertise would be greatly appreciated! Thanks in advance for taking a look!

clydemcqueen, 2025-09-17:
On it

clydemcqueen, 2025-09-17:
OK, I think I found the problem.

### The fix

To use the satellite compass and the [wl_ugps_external_extension](https://github.com/clydemcqueen/wl_ugps_external_extension) you need to set the `--ignore_gps` option when configuring the WL UGPS extension.

This is the configuration you'll get if you accept the defaults:
~~~
{
  "NetworkMode": "host",
  "Env": [
    "UGPS_HOST=http://192.168.2.94",
    "MAVLINK_HOST=http://192.168.2.2:6040",
    "QGC_IP=192.168.2.1",
    "EXTRA_ARGS="
  ]
}
~~~

To use the satellite compass, you must set this configuration:
~~~
{
  "NetworkMode": "host",
  "Env": [
    "UGPS_HOST=http://192.168.2.94",
    "MAVLINK_HOST=http://192.168.2.2:6040",
    "QGC_IP=192.168.2.1",
    "EXTRA_ARGS=--ignore_gps"
  ]
}
~~~

### The symptom (how I found this)

The GPS_INPUT messages have a fix_type that ranges from 1 (no fix) to 2 (2D fix). It never hits 3 (3D fix), which ArduSub requires to trust the GPS readings.

You can see that here -- the green bits never hit 3, even though there is good lat, lon data:

[image]

### Explanation

Basically the G2 box doesn't have a good view of the sky, and it is reporting "no fix" along with the good lat/lon readings coming from the satellite compass. ArduSub will ignore all "no fix" readings.

This is an edge case for the WL UGPS extension, but they support it: You can set the `--ignore_gps` option to tell the WL UGPS extension to always report a fix_type of 3 and at least 6 satellites visible, no matter what.

In any case, the `--ignore_gps` should solve the problem for future transects.

### Remaining puzzle

The fix_type values are bouncing from 0 to 1 and 0 to 2. Even good logs show this bounce from 0 to 3. It looks like every ~18th message reports a fix_type of 0. This could be a bug in the blueos-ugps-extension. It might have no effect, or it might slow down the GPS fix in ArduSub. Hmmm

clydemcqueen, 2025-09-18:
> even though the G2 box shows a GPS lock (based on the lights)

I would expect the 3rd light (GPS) to be blinking, at least for the cases where the fix_type is 1. For fix_type 2 (2D fix) I suppose it _could_ be solid green, as a 2D fix is still a fix even though it isn't good enough for ArduSub.

https://docs.waterlinked.com/underwater-gps/interface/ugps-led/

But... I do find it odd that the 2025_08_13 and 2025_08_21 logs show fix_type values of 0, 1 or 2, but never 3. Is it possible that the G2 box never got a 3D fix? Another puzzle.

In any case, based on my read of the code, setting the `--ignore_gps` option _should_ do the right thing.

> we still need to explicitly tell ArduSub to ignore those GPS readings

Small correction: You need to explicitly tell blueos-ugps-extension, not ArduSub.

By the way, the option name `--ignore_gps` isn't super helpful, it is a bit of a hack. The code should accept the lat/lon values, but overwrite the fix_type and num_satellites values.

> Would it make sense to set --ignore_gps directly in that extension?

That's what you are doing by adding `EXTRA_ARGS=--ignore_gps` to the configuration of blueos-ugps-extension.

> Is there a reason that isn’t the default behavior?

Good question; I would argue for a "make ArduSub happy" mode like this:
- if lat/lon are anything reasonable, then send fix_type=3 and num_satellites=6
- if lat/lon are obviously wrong (e.g., 0/0), then send fix_type=0.

I suppose the WL team is trying to support non-ArduSub use cases as well.

> We’ll make that change in the extension before our upcoming San Juan surveys

I'd love to see it working before you head up there! Keep me posted

m-h-williams, 2025-09-18:
Great, thanks again, Clyde! We'll let you know how it goes!